andreisperid/E-TKT #20: On iOS mobile safari the UI is unusable

sabeechen, 2023-02-17:
On mobile safari on iOS, and no other browser, the 'user-select: none' CSS attribute is interpreted as disabling an element completely.  This makes the UI unusable for the device, as text can not be entered into the label textbox and emoji buttons can't be clicked.

I'll make a PR that fixes this.

andreisperid, 2023-03-20:
@sabeechen I've just fixed a small CSS issue where the button labels became selectable (0869f2f).

I've tested on safari (macOS) and it seemed ok, but didn't on iPhone as I don't have one. Can you please check if its alright?

sabeechen, 2023-03-30:
Yup looks like its still fixed, sorry it took a while to get back to you.

As an aside, I've started using browserstack to test websites on different platforms and found it very helpful for problems like this.  They let you [use it for free](https://www.browserstack.com/open-source) if you maintain an open source repo, I think your repo would qualify.

closed 2023-03-30 by sabeechen

andreisperid, 2023-03-30:
Very cool tip, thank you!

PR #21 (2023-02-25): Make input elements selectable for browsers that care about user-selectable CSS attributes
Fixes #20 on safari mobile, and as far as I can tell has no effect on any other browsers or platforms.