andreisperid/E-TKT #29: J_top overlapping features

sedorr, 2023-04-18:
On the J_top-Part there is a screw hole hitting a countersink.
[image]
[image]

andreisperid, 2023-04-18:
Yes, I have noticed that!

It is not ideal, but the display holder has a little chamfer to avoid interference:

[image]

I decided to keep it that way to avoid major redesign, while keeping the structural capability for the display holder. I could fill in the gap, but it would be so thin it wouldn't make a relevant difference.

andreisperid, 2023-04-19:
Fixed on 4d45f6b, updated STLs, STEPs, BOM and assembly instructions.

Thanks!

closed 2023-04-19 by andreisperid