andreisperid/E-TKT #30: Suggestion: Group and Lable your assemblies

sedorr, 2023-04-18:
It would be cool if you could Group and Lable your assemblies to make navigating the STEP file easier, as currently it's kind of hard to find anything since stuff like the Electronic and mechanic assembiles aren't seperated and all parts are simply named brep.

[image]

andreisperid, 2023-04-18:
Can you please help with that @sedorr ? It would be of great value!

sedorr, 2023-04-18:
I can try, but it would probably be much more helpful to do that in the native file + I don't really know what all the components and surfaces are.

andreisperid, 2023-04-18:
I have exported that STEP from Rhino, which works mostly with polysurfaces, groups and layers.

[image]

When I import it, it goes without problem and assembles polysurfaces, groups and layers. Do you have any clue if there are any import settings that might try to emulate that on Inventor? Maybe a preset for Rhino?

sedorr, 2023-04-18:
[image]

I tried it in Creo as well and it doesn't work.  I have a feeling that Rhino is diong the weird things since there are some parts (That I assume you imported) that are named.

[image]

Maybe try exporting your step files with a [different standard set](https://docs.mcneel.com/rhino/5/help/en-us/fileio/step_stp_import_export.htm), like AP214AutomotiveDesign.

andreisperid, 2023-04-18:
That's exactly the standard that I have used. My bet is that the Rhino's taxonomy does not translate to other software.

What about other formats, such as IGES? Any idea of a standard that might work for you?

sedorr, 2023-04-18:
Maybe just export all your printable parts as their own files? that would also be useful for some slicers, like PrusaSlicer, that take step directly.

andreisperid, 2023-04-19:
Hi @sedorr, I've just uploaded separated STEP files in 2c0aaad

closed 2023-04-19 by andreisperid