andreisperid/E-TKT #31: Suggestion: Post repair information on iFixit

sedorr, 2023-04-18:
Maybe consider posting stuff (Like a Teardown) on iFixit. I already created a wiki and a guide as an example for you: [Link](https://www.ifixit.com/Device/E-Tkt_Label_Printer).

andreisperid, 2023-04-18:
Cool idea, and thanks for putting the page on! I hope E-TKT has a good level of reparability 😄

Maybe the assembly instructions on reverse could characterize as a teardown. Overall, I'm just a bit afraid of spreading the information all around, it makes it hard to update. Hackaday was a strech by itself, but now I'm trying to center everything here at GitHub.

Anyways, I will add a page there for the assembly instructions, that directs to the [E-TKT documentation](https://andreisperid.github.io/E-TKT/) 😉

sedorr, 2023-04-18:
Btw. you can create a page on printables without actually uploading the 3d files. you can just post pictures and a placeholder file and then refer to your github, like [Orbion](https://www.printables.com/de/model/132522-orbion-3d-space-mouse-diy/comments) does. This will give you a bit more reach and a nice place to collect makes that you don't have to update for every minor change.

andreisperid, 2023-04-19:
That's a great idea, I'm definitely looking forward to see more makes out there ;)