andreisperid/E-TKT #34: KiCad local library

andreisperid, 2023-04-20:
I've used symbols, components and footprints installed on my local global library.

To facilitate collaboration and further iterations, I will try to bring this library locally to the project's KiCad directory.

Knochi, 2023-04-21:
Since version 6, symbols are embedded into the KiCAD project, so you don‘t need to. 
It loaded perfectly fine on my site.

https://www.kicad.org/blog/2021/12/KiCad-6.0.0-Release/

> KiCad 6.0 also features a brand-new  schematic and symbol library file format, based on the format used for KiCad board and footprint files. This new format enables long-desired features such as embedding symbols used in a schematic directly in the schematic file, so that cache libraries are no longer needed

andreisperid, 2023-04-21:
Nice, but what about the components, footprints and 3d models?

Knochi, 2023-04-21:
For footprints it’s the same.
For 3D models you can use the plugin „Archive 3D Models“

andreisperid, 2023-04-28:
So I think its alright, thanks @Knochi !

closed 2023-04-28 by andreisperid