andreisperid/E-TKT #35: Custom daisy wheels

andreisperid, 2023-04-24:
I have experimented printing @Knochi parametric design for the custom daisy wheels.

Keep in mind I'm just beginning at 3d printing with resin, there are among my first dozens of pieces. If anyone has experience with it, might be super useful.

[image]
[image]
[image]

Good news:
- the resin wheel characters are able to emboss the label (manually for now, using a plier since the model is not mountable into the device right now)
- it is pretty fast to print both bottom and top parts together (below 2h)

Bad news:
- I have used Anycubic's both Water-Washable Plus and Eco resins, and both are super brittle during the removal of supports. So a few teeth have gone already during the process.
- lets hope the actual embossing with the device is less harsh than the support removal, but I think it will be worse, meaning that probably the wheel will not stand the use with current settings.

----

What I have experimented with:
- **materials:** Eco resin is too oily, I didn't like both the result and process. The Water-Washable: much better, still as brittle as the first;
- **models angle to the printing base:** the bottom part works well horizontally, but the upper gets detached from the supports. I've managed to tilt it +- 20 degrees to lessen the adhesion area to the FEP. The bad news is that there was a subtle distortion on the final piece;
- **supports:** I have tried diminishing the intersection of supports to the models, but it ended up detaching them both. In the end, I have kept the default light settings for the Anycubic slicer, but increased the number of supports to avoid object distortion.

I have changed a few things on the model too:
- overall clearance between upper and bottom part
- increased the text inset height and lateral expansion on the upper wheel
- to be done: modify the bottom part with a coupling the daisywheel to the wheel cube so it can be tested with the real device (might need a custom wheel cube).

----

So, I think there might be at least two ways to tackle these issues:
1 - better suited material: I've seen [people recommend](https://youtu.be/gb_I-9UZem0?t=1590) mixing Siraya Tech's Blu and Tenacious to achieve a semi-flexible result. 
2 - different design: to try not forcing a design that was meant for thermoplastics into resin. Maybe thickening the arms, for instance.

andreisperid, 2023-05-04:
Recently I disassembled my parents' old Olivetti type writer that uses a daisy wheel. @Knochi 

It has a small V notch to help with the characters alignment, which happens automatically when the force is applied:

[image]

The MOTEX wheel does not have it, but with custom wheels it might be possible. Then, the P_press can have the counterpart.

Another interesting feature is this flexible film that holds all the teeth together:

[image]

Might help with the overall strength ;)

Knochi, 2023-05-04:
I will see if I can apply the V-alignment feature. For the flexible film, if it’s just a sheet metal on top of the arms, that would be difficult. But in the scad file there is a parameter to define the size of the inner, solid part. That basically defines the length of the arms. 
One could do a FEM simulation of the ATMs but in the end it’s try and error anyway. 
I see if I can do a benchmarking wheel with different parameters for each arm.

andreisperid, 2023-05-04:
From what I remember, Olivetti's daisywheels are made of some kind of composite plastic

[image]

From the bottom it is possible to see that perhaps it is not the flection of the teeth that make up for the flexibility, but that film itself. Of course, the type writer application is a bit different, because to transfer the tape pigment to the paper, a fast "hammering" is enough. 

In the E-TKT's we are currently pressing to deform the tape, what creates the possibility of horizontal slipping.

But, on the other hand, if we have the flection enabled by that film, we could try increasing the teeth vertical thickness!

andreisperid, 2023-05-09:
Just dropping an idea that I had:

Instead of reproducing the design of the injected plastic daisywheel teeth, there could be some kind of spring to improve the flexibility between the core and the tips.

[image]