andreisperid/E-TKT #38: Cancel label during print

andreisperid, 2023-04-25:
Now that we are just about to allow longer labels to be embossed, I got thinking that we might want to cancel it during print.

A flow would be something like this:
1 - user writes the label and hits print
2 - user desires to cancel, which happens after a modal confirmation (just in case)
3 - the device receives the instruction and finishes the process of embossing up to 1 letter
4 - the device feeds once, then cut
5 - idle and ready to receive new instructions

sabeechen, 2023-04-29:
My personal primary use case for wanting to cancel a print is when I mess something up (like forget to feed in tape, or set incorrect alignment values) and don't want the machine to just grind against itself for a while, in which case having a feed and cut defeats that objective somewhat.

The cancel your proposing (finish letter + feed + cut) is however probably more practically useful for most situations, and having an extra button just do do an "emergency stop" might clutter the UI.  Especially considering I can just push the physical reset button on the device when I mess something up.

I'd like to also propose then that this "controlled cancel":
 - Plays a sad short song as soon as its recieved.
 - Gets some indication on the oled, eg "Canceling label..."

Some care would need to be taken in the firmware for how the cancel is implemented, to ensure it doesn't cancel reboots, parameter saves, feeds, cuts, etc.  It should probably only ever cancel a normal label print.

andreisperid, 2023-05-01:
Good points.

From an interface point of view, I'd use a poka-yoke approach, meaning we shouldn't even be able to cancel what we shouldn't. Together with keeping all other buttons disabled during the cancel procedure, this might eliminate the problem of messing with reboots, saves, etc.

What we want to cancel (and only via web app, meaning it is a "soft cancel"):
- **printing label**: here we might want to finish the current letter, feed (maybe isn't needed at all) and cut (this is needed);
- **testing align/strength**: only stop where it is might do the trick, since its a testing procedure;

So, for each of them, when "in progress" there should appear an nested/overlaid button to the side of the button you used to start the command.

This controlled cancel procedure seems nice! We can use a sad trombone melody and sad emoji :P