andreisperid/E-TKT #39: Synchronize label job through different devices

andreisperid, 2023-04-25:
Right now, if a label is sent to be printed in a device, and then a second device connects to the machine, the latter progress is shown based on whatever is written on the second input label (not the actual label being printed).

For instance, below a second device which had nothing written, while a printing was already been started prior. It misleads the user as if that placeholder text was being printed:
[image]

Any device should show the actual characters that are being printed, with its respective progress.

For that, perhaps the device should advertise the current label.

sabeechen, 2023-04-29:
I 100% agree, I can take this on as its something I've been planning to do ever since I json-ified the webserver.  Should be as simple as returning the current label as part of the status request and accurately rendering it in the web-app.  I'll also make sure it doesn't overwrite a label you're editing if a print starts in the middle of that.

sabeechen, 2023-04-29:
I've got a PR ready to go for this [in a branch](https://github.com/sabeechen/E-TKT/tree/label-webapp-sync), but it depends on the changes in my [OTA PR](https://github.com/andreisperid/E-TKT/pull/41), so I'll wait until that PR is reviewed (or rejected!) to request review.

andreisperid, 2023-05-01:
Amazing, thank you again @sabeechen ! 

And it is really nice that you made this parallel component to keep the previously written text 👏 #43

closed 2023-05-01 by andreisperid