andreisperid/E-TKT #44: Label cutting is erratic

andreisperid, 2023-05-01:
One of the things that never made me happy with the device is the cutting performance. Its still quite random.

What happens now:
- the machine presses three times to make sure the blade is able to pierce through both the label and liner;
- sometimes the labels are perfectly cut, sometimes not;
- eventually, there are white marks on the extremities, due to excessive press (energy is applied in the wrong place?);

Hypothesis:
- it seems that depending on the first cut, the label tilts a bit creating a better clearance for the next cut;
- maybe the label thickness varies, impacting performance;
- the blade gets dull (or comes with irregular sharpness?);
- the press might be hitting erratically, missing the ideal daisywheel spot;
- the homing might be imprecise, leading to the item before;
- the current servo might be enough for the characters, but not for cutting;

The ideal scenario would be:
- fully cut the label, but leave just a tiny bit of liner uncut to make sure the label isn't carried by the daisywheel and get entangled;
- prevent damaging the tape (white marks);
- make sure the little tab (double cut) is functional to help removing the liner, which means fully cutting the external (except tiny bit of liner), but for the inner, only cut the label, but not the liner;
- press only once to cut;

sabeechen, 2023-05-03:
I have a few ideas, though I also wonder if just one cut might be enough.  I've grown rather fond of the little pre-cut tab that sometimes forms, it makes getting the backing off *really really cheap* label tape easier.  I suspect the mechanics of the cutting is actually very chaotic.

My best guess about what causes the displacement during sequential cuts that should be in the same place is slack in the channel that the label runs through, since the label tape wants to curl after it comes off the tape.  I'll attempt to describe my theory with crude drawings:
[image]
Before a cut, viewed from the side, the tape forms one solid arc back toward the feeder.  After the first press on the cutter the kink allows the tape to form two more shallow arcs, which covers the same horizontal distance with less tape, causing a slight shift where the next cut takes places.  The cut itself will also definitely cause the tape to stretch a little as the tape squeezes out the sides of the blade, also causing some displacement.  I suspect that the tape doesn't always have a lot of slack/arc, which is what makes the subsequent cuts only sometimes land on a different spot.

Some ideas I have to have solve this, though they'd all take some experimentation to get right or even prove worthwhile:
1. Instead of fully returning the press to its rest position for each cut, only back off slightly so some tension is left on the tape keeping it in position.  Subsequent cuts would then kind of "chew" through the tape without fully releasing it.
2. Only do one cut, but while the press is still raised pull back slightly with the feed stepper.  Pulling back might help encourage the tape to tear in the right place without too much collateral damage to the edges.  It might even be worthwhile to move the tape back and forth a little.
3. If the displacement in subsequent cuts could be characterized somehow, the feed motor could feed/retract the correct amount between cuts to get the tape to land in the right spot.

All of these could even be combined together to make the E-TKT do a grindy little dance during a cut.  Though it may also be the case I'm way off base and none of this would help.

andreisperid, 2023-05-03:
Great observation and ideas. I like very much the bite-chew, and also the bite-tear.

To avoid missing steps that might occur in the latter, maybe we can even reduce the acceleration to raise the torque.