andreisperid/E-TKT #54: Add support for Trinamic drivers

sedorr, 2023-05-07:
Maybe add support for StallGuard4 and StealthChop2 via the TMC2209. It should be relatively easy since you already use the standard driver modules used in 3d printers. 
If you don’t need stall detection the TMC2208 should even be a drop in replacement for the A4988 (in legacy mode at least as far as I know).

andreisperid, 2023-05-08:
Pretty cool, nice idea. I've been interested on experimenting with the TMC2209 for sometime.

By quickly looking in the [datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_datasheet_rev1.08.pdf), it seems that an UART port is needed to fully configure it, so maybe we can implement it with the next version.

Didn't know, but there is a way to use a sensorless homing, but unfortunately it is not compatible with the E-TKT. It needs a physical stop to detect home.

sedorr, 2023-05-08:
Maybe even consider doing away with the stepstick boards and solder a TMC2209-LA (or two) to the board. 

If you don’t need stall detection the TMC2208 should be a drop in replacement for the A4988 (in legacy mode at least as far as I know). On the 2208 the UART is used for advanced stuff like full 256 microstep control and hybrid switching betwean stealthchop and spreadcycle mode. But you don't need it if you only need stealthchop and max. 16 microsteps you can get away without the UART.

sabeechen, 2023-06-23:
I tried swapping in a TMC2208, it is indeed a drop-in replacement for the A4988.  It seems to work the same, except I had to reverse the stepper direction in the firmware.  For anyone else doing it id recommend getting one preconfigured to disable UART mode, since its really difficult to move the config SMD resistor without a microscope, especially if the pin headers are already soldered.

The biggest difference is that the motor is nearly silent when moving (except for plastic rubbing on plastic) and the motor whine when holding position is gone.  I rather like it.